cfelton/rhea #5: Can't make blinky work on Mojo v3.

gbin, 2015-12-10:
I am very new to the FPGA world so I might have done something stupid :)

My source file is:
```python
from myhdl import always, always_comb, always_seq, Signal, ResetSignal, toVerilog, toVHDL, delay, traceSignals, Simulation, now, intbv, concat

from pprint import pprint
import rhea.build as build
from rhea.build.boards import get_board

def blinky(led, clock, reset=None):
  assert len(led) >= 2

  nled = len(led)
  maxcnt = int(clock.frequency)
  cnt = Signal(intbv(0,min=0,max=maxcnt))
  toggle = Signal(bool(0))

  @always_seq(clock.posedge, reset=reset)
  def rtl():
    if cnt == maxcnt-1:
      cnt.next = 0
      toggle.next = not toggle
    else:
      cnt.next = cnt + 1

  @always_comb
  def rtl_assign():
    led.next[0] = toggle
    led.next[1] = not toggle
    for ii in range(2, nled):
      led.next[ii] = 0

  if reset is None:
    reset = ResetSignal(0, active=0, async=False)

    @always(clock.posedge)
    def rtl_reset():
      reset.next = not reset.active
    g = (rtl, rtl_assign, rtl_reset,)
  else:
    g = (rtl, rtl_assign,)

  return g

def run_mojo():
  brd = get_board('mojo')
  flow = brd.get_flow(top=blinky)
  flow.run()
  info = flow.get_utilization()
  pprint(info)

if __name__ == '__main__':
  run_mojo()
```

my build file:
```bash
#!/bin/bash
. /opt/Xilinx/14.7/ISE_DS/settings64.sh
python chenil.py
promgen -u 0 xilinx/mojov3.bit -b -p bin -w
```

Note: I have tried to generate directly the .bin too by adding the line to the tcl script in ise.py:
`self.tcl_script += 'project set "Create Binary Configuration File" "true" -process "Generate Programming File"\n'`

Then I run mojov3.bin with mojo-loader but nothing happens, the DONE led is not even light up as it does when I load a correct .bin.

cfelton, 2015-12-10:
@gbin there is probably a default value in the bitfile generation that is incorrect for the Mojo.  I don't have a Mojo board, someone else contributed the board definition.  I can try and investigate in the next couple days and/or track down the original contributor and see if there is something missing.

As a debug option, you can try and open the generated project file (.xise) with ISE, run the flow, verify it still fails.  Then try and modify the bit generation settings (probably a pull-up (or not) on done or something) and see if it works.

gbin, 2015-12-10:
So if I open the file verbatim with ISE, compile click on "Generate Programming File", took the resulting bin and mojo-loading the bin, it works !

gbin, 2015-12-10:
The problem is in the tcl script,
adding `self.tcl_script += 'project set "Create Binary Configuration File" "true" -process "Generate Programming File"\n'`
and removing `self.tcl_script += "project set \"FPGA Start-Up Clock\" \"JTAG Clock\" -process \"Generate Programming File\" \n"`

Makes the bin file directly loadable with mojo-loader.

cfelton, 2015-12-31:
@gbin, I added a simple flag to disable the jtag start-up clock.  This should be fixed and work in the base.  I will keep this open until verified on hardware.

cfelton, 2016-06-19:
@gbin, I believe this has been fixed but I don't have the hardware to verify.  I am going to close this let me know if it is a problem in the future.

closed 2016-06-19 by cfelton