cfelton/rhea #17: Long simulation tests

cfelton, 2016-03-13:
Some of the tests in the test suite take considerable time to execute.  Two things need to be done.  First, try and reduce the test, use transactors and subset testing to reduce the test time.  Second, mark the long running "full" tests as `long_test` (or something).  The pytest package has a mechanism to mark certain test and select when these are run.  The "long" tests will not be run normally but they can be run manually for additional test coverage.

Currently, many of these tests are skipped (`pytest.mark.skip(True)`).  These are the tests that need to be dealt with.

forumulator, 2016-03-22:
Created a pull request ( #20 ) initially marking the slowest tests.

cfelton, 2016-03-22:
Thanks for the PR!  I realize I mentioned `long_test` in the description but as discussed in #20 a longer name (short-hand for the pytest descriptor) is preferred.  Thanks for updating the PR to reflect the more descriptive name.

closed 2016-03-26 by cfelton