cfelton/rhea #24: Change GPIO pin assignments to Peripheral based devices vs FPGA based.

Godtec, 2016-04-04:
After running into some small issues converting the de0nano for the de1_SoC board. I noticed that the LT24LCD pins are hard coded into de0nano pin definition file. 

In my opinion, these pins should not be put here. Since these pins are connected to a peripheral board. It should be flexible upon which board you use. 
(Said another way, the GPIO pins should be defined by the peripheral you use, Not the FPGA board)

I am proposing a new file called "lt24lcdpins.py" under the directory rhea-master/rhea/cores/video/lcd...
 this file will linkup with the dedicated pins on the de0nano or de1_SoC GPIO[0/1] pins. 
The user will define which GPIO to use (eg 0 or 1),  in the examples directory under the board
 definitions de1_Soc_lt24lcd.py file, in the examples directory.

 Please let me know your feedback, Or a better location to put this type of file.
Thanks. MikeK

cfelton, 2016-04-04:
The correct method would be to create a board definition specific to the hardware configuration as described in the [documentation](http://rhearay.readthedocs.org/en/latest/build/board_definition.html#extending-definitions).

These hardware configurations are typically specific to an application so it is hard to include definitions for all the combinations, this is something you would have in your external project.  To provide reusability the idea is to create external interface definitions in the base class (as the comments indicate in the source) with some method to map various pins, but the external interfaces are a WIP.

In the package, I can remove the LT24 specific signals and create a custom board definition for the examples but it is also benign to have multiple port definitions (I will probably leave them for the time being).  The generic GPIO expansion headers are in the board definition but they are entered slightly different as one port `gpio` instead of `gpio_0` and `gpio_1` a single `gpio` port provides more flexibility.

In summary:
   1. correct, the base board definition should contain the generic connectors as defined in the board documentation but having specific overlapping pin definitions is benign.
   2. custom configurations should be defined in the user application.
   3. external interfaces are used to provide reusable objects in a board definition and a board definition would only contain common uses.

cfelton, 2016-04-04:
The documentation can be expended (I leave this issue open until it is added) to include some more examples and uses cases.  Example, the docs only show an example how to overwrite the complete default ports:

```python
class MyCustomBoard(Xula2):
    # overriding the default ports, inherit the default
    # clocks.  The default ports in this cause reprsent
    # the various widgets connected to the Xula2+stickit
    default_ports = {
        'leds': dict(pins=('R7', 'R15', 'R16', 'M15',)),
        'btns': dict(pins=('F1', 'F2', 'E1', 'E2',))
    }
```

You could also do something like:

```python
class MyDE1SOCBoard(DE1SOC):
    def __init__(self, lt24_gpio_bank=0):
        if lt24_gpio_bank == 0:
            lt24ports = {}
        else:
            lt24ports = {}
        self.default_ports.update(lt24ports)
```

Godtec, 2016-04-04:
I have completed the lt24LCD for the DE1-Soc, I have made comments in the IRC chat.
Please review.
I just followed the de0nano example, Similar to the #todo comment.

Godtec, 2016-04-04:
I will leave this issue open for now, I will add more code later once I understand your process a little more. (And python) I am going to implement the Video VGA on the DE1_SoC, Since I can FPGA test it.

cfelton, 2016-04-04:
@Godtec my examples are a incomplete might be why they are hard to follow.  Essentially, for each hardware with specific components connected you want to create a new board definition for that hardware configuration.  Because these configurations can be numerous it doesn't make much sense to include them all in `rhea`'s collection of board definitions.  

You can create a pull request for `DE1SOC` at any point.  Small note on naming, see the Python PEP8, the other classes (board definitions) have not used underscores in them we will stick with the all uppercase (since they are all acronyms).  Also, I have been slowly removing the leading '_' in all the filenames in the package, this will be true for the board-definitions.  Your new files can exclude the leading '_'.

Thanks for the feedback and additions soon to come :)

Godtec, 2016-04-04:
So Basically already done. I created a new DE1_SoC pin definitions file and complied the lt24lcd example file. Now the issue is getting my git to work! :) Still new to it.

Now, Since you do not have an explicit VGA example similar to lt24lcd, I am scratching my head on how it works. I may (WILL) have questions. If that's ok. I am referencing the test_vga examples for help.

closed 2016-08-24 by cfelton