cfelton/rhea #34: Would be nice if Clock() type exposed 'clock ticks' to 'simulation ticks' ratio.

NickShaffner, 2016-07-28:
This would be particularly useful for being able to delay a certain number of clock ticks.   Adding a delay() generator method to the clock type would be another means of exposing this.

ex:

```
something.next = False
yield glbl.clock.delay(1)
something.next = True
```

cfelton, 2016-07-30:
@NickShaffner this sounds reasonable, my gut says a slightly different name: `wait` or `pause` possibly, mainly to avoid confusion with `myhdl.delay`.

NickShaffner, 2016-07-30:
I can see that - wait() would work for me. Also about the clock ticks to simulation ticks constant - how about 'ticks'? (since _hticks is used as half ticks in the clock already).

NickShaffner, 2016-08-05:
https://github.com/cfelton/rhea/pull/40 has Clock().ticks,  Will add the delay() bit in over the weekend.

NickShaffner, 2016-08-05:
https://github.com/cfelton/rhea/pull/40

NickShaffner, 2016-08-11:
Added in: https://github.com/cfelton/rhea/pull/40

closed 2016-08-11 by NickShaffner