cfelton/rhea #41: Vivado 2016.2 can't process generated .tcl files with absolute paths or spaces on windows 10

NickShaffner, 2016-08-07:
Generated a simple switches to leds program for zybo, and the following bit of the flow:

>  vivado -mode batch -source Z:\Dropbox\Projects\PU\build\pu.tcl

resulted in the following output:

`
# add_files Z:\Dropbox\Projects\PU\build\pu.v
ERROR: [Vivado 12-172] File or Directory 'Z:DropboxProjectsPuildpu.v' does not exist
`

The offending line of the TCL file:

`
add_files Z:\Dropbox\Projects\PU\build\pu.v
`

The same issue happens for all other paths in the file, and is corrected by changing the paths to forward slashes.  (Which the vivido gui seems to do in it's own 'choose a file' dialogs).

Fix incoming shortly.

cfelton, 2016-08-07:
Thanks for tackling this issue in your current PR!

NickShaffner, 2016-08-07:
Instead of changing these to forward slashes as vivado does, I opted to escape the backslashes - so I don't end up 'second guessing' the os.path.join black box.

NickShaffner, 2016-08-11:
This should be fixed with https://github.com/cfelton/rhea/pull/40 but should also be double checked for the other tools (eg ISE)

NickShaffner, 2016-08-14:
Reopening because this is also the case for ISE (14.7) - fix incoming.

NickShaffner, 2016-08-14:
https://github.com/cfelton/rhea/pull/52 fixes this for ISE 14.7 on Windows 10

NickShaffner, 2016-08-16:
ISE side of this merged with https://github.com/cfelton/rhea/pull/52

closed 2016-08-16 by NickShaffner