cfelton/rhea #47: Would be nice if flow.run() provided board information to the top module for things like ClockManagement

NickShaffner, 2016-08-11:
Suppose I'm building a design like so:

```python
    board = rhea.build.boards.get_board("cmoda7_35t")
    flow = board.get_flow(top=top_cmoda7_35t)
    flow.run()
```

If (for example) I'm using a ClockManagement module, I've currently got to hard-code the vendor string in the top implementation:

```python
@myhdl.block
def top_cmoda7_35t(clock, led, btn, uart_rxd_out, uart_txd_in):

    reset = Reset(0, active=0, async=True)
    clkmgmt = ClockManagement(clock, resetext, output_frequencies=(125e6,))
    clkmgmt.vendor = 'xilinx'
[...]
```

it would be nice if I didn't have to hardcode the `clkmgmt.vendor = 'xilinx'`, but instead had access to the board instance being used to portmap this guy, so I could just pull it from board.vendor. (`clkmgmt.vendor = board.vendor`)  Perhaps board could be mapped into the function call similarly to the way the ports are, so you could instead do:

```python
@myhdl.block
def top_cmoda7_35t(board, clock, led, btn, uart_rxd_out, uart_txd_in):

    reset = Reset(0, active=0, async=True)
    clkmgmt = ClockManagement(clock, resetext, output_frequencies=(125e6,))
    clkmgmt.vendor = board.vendor
[...]
```

Thoughts?

cfelton, 2016-08-14:
@NickShaffner It is the design intent for the board information to be passed to the design, I have not determined if this should be automatic of if the designer needs to include a `board` parameter in their top-level and the `board` passed.

```python
def  top_cmoda7_35t(clock, led, btn, uart_rxd_out, uart_txd_in, board=None):
    clkmgnt.vendor = board.vendor
```

I believe this is the most straightforward option but the `board` parameter will be specific, if some uses `board` as a port it will fail.

NickShaffner, 2016-08-14:
Yeah, I can see the issue there - perhaps something more unlikely to be reused?

`rhea` or `params` perhaps

with a member `board`:

`rhea.board`

(and room for future expansion as well :))