cfelton/rhea #51: ClockManagement's 'enable' parameter is not implemented on Xilinx parts

NickShaffner, 2016-08-14:
Is it me, or is there some Altera favoritism going on in Rhea? :)

ClockManagement's 'enable' signal is not implemented on Xilinx parts, though it is on Altera parts and the Simulator.   This should either be consistent for Xilinx and other brands of parts as well. 

Ways I can think of to do this:
* Adding clock control units to the outputs of the Xilinx MMCM_BASE - would need more than one
* Somehow provide an error when enable is assigned on non supported hardware.
* (personal preference) Removing the 'enable' signal altogether 

Thoughts?

cfelton, 2016-08-16:
@NickShaffner I agree it should be consistent across all vendors, it might be reasonable to remove it all together - will need to think about it some.

cfelton, 2016-08-24:
Think about this, `enable` will be an optional port, kinda like `reset` in some cases.  The default will be `None` (that means it will need to move to the end of the port list).