cfelton/rhea #53: Symbol 'MMCME2_BASE' is not  supported in target 'spartan6'.

NickShaffner, 2016-08-14:
Would be nice to have clock management primitives for the spartan 6 series, It seems MMCME2_BASE is 7 series logic and above.

NickShaffner, 2016-08-16:
@cfelton What about moving away from directly modeling clock managers and clocking resources (which seem pretty diverse across devices in their capabilities and factoring).  Have you considered adding the abstraction directly to clock?

```python
clock           # example hardware clock declared in board file - 12Mhz

my_derived_clock = Clock(0, frequency=100e6, driver=clock)
```

I could see potentially adding `phase` and `enable` parameters as well.   I imaging after being declared as suck, clock resources could be collected in a global registry and some magic could occur to generate the appropriate chains of clock resources using vendor and part specific hardware. (Generating errors when such hardware isn't available)

cfelton, 2016-08-24:
@NickShaffner Yes, we can and should add the spartan DCM and some of the other stuff we discussed (like removing enable since not all clock managers have it).

As far as the second goes, I think that would be a possibility but much farther in the future, we would need a generic'ish things like the vendor primitives under the hood this should work to such a solution.  We should create a PEP/MEP type thing to fully outline such feature requests.  We can use the issues mark them as enhancements but require the same thoroughness as a PEP.

NickShaffner, 2016-08-24:
> We should create a PEP/MEP type thing to fully outline such feature requests. We can use the issues mark them as enhancements but require the same thoroughness as a PEP.

Is it too early for a PEP style feature request system?   Part of the reason I added the suggestion about the clock(driver=) enhancement was to do a little 'exploratory prodding' to learn more about the 'architectural style' and boundaries of the project.