cfelton/rhea #56: instructions: xula_blinky_host.py on raspberry pi 3B

hstarmans, 2016-11-17:
Dear all,

I am trying to get xula_blinky_host.py working with a; Raspberry Pi 3B, Xula2-lx25, StickIt!-LedDigits and the Stickit!-MB v4.
I have inserted the LedDigits in PM2. I am able to to run the LedDigitsTest, after changing the LedDigitsTest.UCF using xsconnect.
I have verified the UART pins are working, by connecting chan 14 and 15 on the stickit board.
I communicated via minicom and checked if I could receive echoes.
I am able to create a bitstream from xula_blinky_host. If I upload this bitstream; nothing happens.
The LedDigits module is off. This is not what i expect.
I thought data would be written via uart to the xula2 and this would appear on the leddigit board.

Do the authors, @cfelton and @forumulator , have any advice  on how to proceed!? Which settings should I use in minicom!? Should I try the example with a raspberry pi 2? What setup was used to verify the code!?

Notes:
I use ubuntu 16.04.1, have a jumper on 5V-PWR and a jumper on GPIO-5V.
I did not change the EEPROM as this does not seems to be possible on the Raspberry Pi 3. 

Keep up the good work,

Rik

cfelton, 2016-11-18:
You need to create a ["packet"](https://github.com/cfelton/rhea/blob/master/rhea/cores/memmap/command_bridge.py#L16) to send down to toggle the lights.  The simple UART is typically configured for 115200.  It has been awhile since I have looked at it, let me see if I can get some hardware setup this weekend and test.

hstarmans, 2016-11-19:
Thank you for your reply,  I guess I need something along the lines of;
'''python
import serial, binascii
ser=serial.Serial("/dev/ttyS0") 
ser.baudrate=11520              
hexdata='DE'+'address bytes'*4+'lengt of data'+'CA'+'data byte'*4
data=binascii.unhexlify(hexdata)
ser.write(data)
ser.close()
'''

cfelton, 2016-11-23:
[This is the code](https://github.com/cfelton/rhea/blob/master/rhea/utils/command_packet.py) I use to generate the packets.  I have not tested/used from a RPi (others have).  I am not sure what code they used on the RPi.

hstarmans, 2016-11-24:
Dear Felton,
Thank you for your reply. I am still unable to get it to work. I have used the following code to test it:
I don't receive any replies from the xula2 or do something visually on the LedDigitsTest.
"""python
from rhea.utils import CommandPacket
# connect to serial port
ser=serial.Serial("/dev/ttyAMA0",timeout=2) 
ser.baudrate=115200
# write value 100 to address 31
wcommand=CommandPacket(rnw=False,address=31,vals=[100])
ser.write(wcommand.rawbytes)
# Nothing happens on the board            
# read address 31
rcommand=CommandPacket(rnw=True,address=31,vals=None)
ser.write(rcommand.rawbytes)
# read one byte from the serial buffer
ser.read(1) 
# --> no value is received ; does not seem to work
"""

hstarmans, 2017-01-10:
I have spotted one bug:
```python
 # create the UART instance.
    uart_inst = uartlite(
        glbl, uart_fifo, serial_in=bcm14_txd, serial_out=bcm15_rxd,
        fifosize=4
)
```
Should be:
```python
 # create the UART instance.
    uart_inst = uartlite(
        glbl, uart_fifo, serial_in=bcm15_rxd, serial_out=bcm14_txd,
        fifosize=4
)
```
I have tested the new code with: 
```python
from __future__ import print_function, division

import myhdl
from myhdl import Signal, intbv, instance, delay, StopSimulation, ResetSignal

from rhea.system import Global, Clock, Reset
from rhea.models.uart import UARTModel
from rhea.utils.test import run_testbench, tb_args, tb_default_args, tb_convert
from rhea.utils import CommandPacket
from rhea.utils.command_packet import PACKET_LENGTH

from xula_blinky_host import xula2_blinky_host

def test_ibh(args=None):
    args = tb_default_args(args)
    numbytes = 13
    reset= ResetSignal(0, active=0, async=True) # NOT USED!
    clock = Clock(0, frequency=12e6)
    glbl = Global(clock, None)
    led = Signal(intbv(0)[8:])
    pmod = Signal(intbv(0)[8:])
    uart_tx = Signal(bool(0))
    uart_rx = Signal(bool(0))
    uart_dtr = Signal(bool(0)) # NOT USED
    uart_rts = Signal(bool(0)) # NOT USED
    uartmdl = UARTModel()

    @myhdl.block
    def bench_ibh():
        tbclk = clock.gen()
        tbmdl = uartmdl.process(glbl, uart_tx, uart_rx)
        tbdut = xula2_blinky_host(clock, led, uart_tx, uart_rx)
        
        @instance
        def tbstim():
            yield delay(1000)
            
            # send a write
            pkt = CommandPacket(False, address=0x20, vals=[0xFF])
            for bb in pkt.rawbytes:
                uartmdl.write(bb)
            waitticks = int((1/115200.) / 1e-9) * 10 * 28 
            yield delay(waitticks)  #  code breaks here ==> assertion error
            timeout = 100
            # get the response packet
            for ii in range(PACKET_LENGTH):
                rb = uartmdl.read()
                while rb is None and timeout > 0:
                    yield clock.posedge
                    rb = uartmdl.read()
                    timeout -= 1
                if rb is None:
                    raise TimeoutError
            yield delay(waitticks) 
            # the last byte should be the byte written
            assert rb == 0xFF

            yield delay(1000)
            raise StopSimulation

        return tbclk, tbmdl, tbdut, tbstim

    run_testbench(bench_ibh, args=args)
    inst = xula2_blinky_host(
        clock, led,
        uart_tx, uart_rx
    )
    tb_convert(inst)

if __name__ == '__main__':
    test_ibh(tb_args())
```
Note that, I have removed the reset!!  I have removed the reset pin as this was also not used in the icestick example.
The code still does not work on the Raspberry pi.

hstarmans, 2017-01-11:
@cfelton ; shall I change xula_blinky_host and add my rasptest and test_blinky_host_xula!?

cfelton, 2017-01-12:
@hstarmans can you link the documentation that shows the names being flipped?  It has been awhile and I don't recall if I was trying to standardize the UART signals always being from a certain perspective.

Go ahead and create a PR and I will review it.